a-poorva/Design-Project #6: Software Review

a-poorva, 2018-11-06:
I'm having some problems with

1. Bouncing 
I'm not sure what my program isn't accounting for. 

2. Brightness changes
The LED isn't really getting any brighter, or less? Not sure what to change in my program. Should I just assign specific values for each level of brightness and remove the if statement?

mlp6, 2018-11-06:
It looks like you are using `digitalRead()` instead of `analogWrite()` for the PWM output in some of your case statements; am I misreading that?

W/r to the bouncing, what happens if you modulate the debounce delay?

mlp6, 2018-11-24:
I am not sure I totally follow how your debouncing logic should be working, but one thing I would be careful with with this line:
https://github.com/a-poorva/Design-Project/blob/8bf33d13c4cfb9eff1144b0b43db92085eee905f/Software/LEDv3.ino#L69

`ledState` is declared as an `int`; you are using it in that line more like a `bool`; can you declare it as a `bool` instead so that you can just toggle the state?  Also, I'm not a huge fan of assigning the value by flipping logical state; it would be easier to follow if you explicitly set and tested "on" and "off" states for the button.

a-poorva, 2018-12-02:
Hi Dr. Palmeri,

So I fixed my code a little, and while it changes states smoothly, the whole thing just runs on a loop continuously, not sure why. Would you mind taking a look at my code?

mlp6, 2018-12-03:
Your `loop()` function doesn't seem to ever call `button_press()`.

closed 2018-12-05 by a-poorva