a-poorva/Design-Project #7: PCB design

a-poorva, 2018-11-08:
1. I still have to assign this to Matt Brown and Dr. Richardson, since I sent them collaborator requests only last night/this morning. I'll add them later.
2. I started on my PCB design by adding polygon pores and then routing, but some of my connections seemed to disappear when I did that. Brinnae suggested routing and then adding polygon pores, and ripping up the ground connections? I'm a little confused about that, as well as as to how to proceed beyond that (in terms of adding holes, vias, etc.)
3. Can I cut out all that extra space when I create my polygons?

a-poorva, 2018-11-09:
When I showed my design to Matt Brown today, he mentioned that part of the reason why my battery is just sort of floating around is because I didn't connect the Vcc in my schematic, though I labelled it?? Besides drawing connections between them, is there a more elegant way of approaching this problem?

a-poorva, 2018-11-09:
So I tried adding the source with the net tool, and I tried to reroute it but I noticed a problem(?) I saw last time as well: 

1. My switch only has one connection?
2. The positive terminal of my battery doesn't seem to be connected to anything, but both connections seem to arise from the ground terminal. 

Not sure what the issue is.

mlp6, 2018-11-09:
Will take a look in the morning.

a-poorva, 2018-11-09:
Uploaded a new version! Does this look a little better?

sphaerobolus, 2018-11-12:
Rather than creating new files for each version, take advantage of the version control system: Just upload the new version. The old version is still available as an old version of the same file. 

Bypass capacitor C1 would be better between the regulator output and ground. It's not doing much good as used here, and might be doing harm. 

Change the 100k current limiting resistor to a real value. 

If you're doing this as a shield, do you even need the regulator? You could use Arduino's the on-board regulator. You pass through the Arduino's regulator if you use Vin stead of 5V as your supply.

a-poorva, 2018-11-13:
Thanks for your help, Matt! I've uploaded a new one (I have been uploading all my files so far, using the 'Upload files' button! Is that not the right way to do this?). Hope this one is better.

sphaerobolus, 2018-11-14:
I didn't use the right terminology when I suggested not creating new files. if you're using the desktop git client, the things you want to be doing are "commit" to save a new version of an existing file and "push" to put the file on the github server. If you're using the web client, I'm not sure what the best process is. You could edit the file, then paste in the new version and save it. You definitely don't want a bunch of files named "abcdv1.brd" and "abcdv2.brd". The git ecosystem takes care of the versions, you just commit new versions of existing files. 

The board is looking good! Now I think all you need is to increase the width of traces. You can probably get away with 20mil, so go for that.

a-poorva, 2018-11-16:
Right. I'll be sure to look up how exactly to do that. 

Thanks for your help!

closed 2018-11-16 by a-poorva