a-poorva/Design-Project #8: Header pins

a-poorva, 2018-11-16:
I just realised that I used the arduino board in my PCB instead of header pins. Is this the right way to do this? I'm not sure I understand the difference here.

sphaerobolus, 2018-11-17:
If you want this to be a shield for your Uno, you need to line up your connections to the Uno. To do this, you placed the Uno board as a component. If I understand you correctly, you might just be confused over semantics. Are you placing the Uno on your board? Or your board on the Uno? Doesn't matter, the holes still line up. 

After fabricating the board, you'll put header pins in all those holes around the perimeter, then plug your board into the Uno. 

Did I answer the right question?

a-poorva, 2018-11-17:
Right, so it doesn't matter, essentially, whether you use header pins or the actual board in your design, as long as the pins line up?

mlp6, 2018-11-17:
Yep!

closed 2018-11-17 by a-poorva