a-poorva/Design-Project #9: Mechanical design

a-poorva, 2018-11-18:
Please see the uploaded designs for reference. 

I'm having a few problems with the following: 

1. The arduino seems to be underneath the whole cylinder? I'm not sure why that's happening, or if I"m misinterpreting the design. 
2. I wanted to have the lid of the cylinder screw on. I'm not sure if I've done the threading correctly. 
3. The holes I've made in the lid seem to vanish whenever I try to dimension them - is this because I've made them incorrectly, or is there some other reason? 

Any other feedback would be great!

ps178, 2018-11-19:
Could you upload the .f3d version of your CAD design.
You also need to upload the 2D rendering of your design. You can find this under MODEL -> drawing

a-poorva, 2018-11-19:
Done and done!

ps178, 2018-11-20:
1) Yes, your PCB is underneath the cylinder. This is because you sketched the cylinder on the same plane as where you placed the board. Usually, you would create you casing based on the PCB & arduino dimensions and then you can assemble with the 3d PCB rendering to make sure everything fits together. Does that make sense? I think you can create you circle first and then insert your PCB.

2) I think the threads look good. You have them going in different directions so they should fir together given their thread size and count are the same.

3) It looks like you made two holes with 10 mm diameter. Is that not what you were trying to do? Also, instead of using the 'holes' command you can also sketch a circle and the extrude in the opposite direction to carve out the circle onto another surface.

4) The holes for the LED & button or the holes for mounting the arduino?

sphaerobolus, 2018-11-23:
I'm not sure you'll be able to print a thread suitable for a project like this. Our FDM printers are pretty low resolution. (An SLA printer has hope of printing some threads.) 

You have two options: You can print a hole suitable for a wood screw or self-tapping screw. This screw would form its own threads in the plastic the first time you put it in. 

Better, I think, are heat-set thread inserts. You form a hole with a little bit of taper (a/k/a draft angle). The top of the hole is just big enough for the end of the insert to fit in "dry". After dry-fitting, you use a heat source, such as a soldering iron, to melt the insert into the plastic. You can google the process. The inserts I have are from this page: 

https://www.mcmaster.com/catalog/124/3395

We have the tapered inserts in 2-56, 3-56, 4-40, and 6-32.

closed 2018-12-05 by a-poorva