a-poorva/Design-Project #12: Sliding Door Idea

a-poorva, 2018-11-26:
Would this be possible to print such that the piece that slides in moves smoothly?

mlp6, 2018-11-26:
Can this be printed... yes, especially if the overhanging notch isn't too much to require supports to be printed that could be really tricky to get out and remain smooth... but "smooth" operation will really be dependent on the quality of the print and making sure that you allow for some tolerance in that sliding piece.

@sphaerobolus can comment on the specifics of the printers that he runs to see what he would recommend.  We've done sliding access in the past; most have worked.

ericsrichardson, 2018-11-26:
One thought on sliding doors - it's often helpful to round off the two corners on the leading edge of the sliding door (if I'm understanding what you want to do correctly) so that when you slide the door into the slots, the corners don't bind inside the slots.

sphaerobolus, 2018-11-26:
Also, watch for clearances. The printers overprint, meaning the parts can end up being slightly larger than you designed. If I have two parts that I need to nest, I usually make a difference of at least 0.020". Say my nominal dimension is 0.5". I might make one part 0.49" (or maybe smaller) and the other 0.51" (or maybe larger).

closed 2018-12-05 by a-poorva